ioBroker/ioBroker.s7 #7: Bump node-snap7 version

mathiask88, 2017-07-11:
Since version 0.4.0 node-snap7 supports a set of platforms and node versions as prebuilt binaries that are automatically downloaded on installation. Compilation is now the fallback. Maybe this is interesting for you.

Apollon77, 2017-07-12:
Is there any breaking change to 0.2.x?! Do you know that?
If not we can upgrade and test (or because of "^0.2.3" most users that install now should get a current version automatically :)

mathiask88, 2017-07-12:
@Apollon77 No there should be no breaking changes and no the users won't get it automatically. You can read about that magic caret with leading zero [here](https://docs.npmjs.com/misc/semver#caret-ranges-123-025-004) :)

Apollon77, 2017-07-12:
@mathiask88 Thank you, learned something :-)

one question: on github page you write that node 4.x is needed at minimum ... is this new or was it already with 0.2.3? Because upgrade would change the minimum node requirements and is kind of "breaking"

Apollon77, 2017-07-12:
Sorry, ignoe question, mis-read a line in your README :-)) Everything great

mathiask88, 2017-07-12:
I can't tell you if it works with node 0.8, because I dropped it recently from the CI config, but I hope noone is using 0.8 anymore :S

Apollon77, 2017-07-12:
We start with 0.10 in testing, but all users are on 0.12 at least (most on 4.x I think)

fixed by https://github.com/ioBroker/ioBroker.s7/commit/fc03f3924847103ad5853ae39b350d8372d598b9

Testing thread: http://forum.iobroker.net/viewtopic.php?f=36&t=7209

closed 2017-07-12 by Apollon77