ioBroker/ioBroker.s7 #13: REAL zahlen werden scheinbar duch zufällige Zahlen vom s7.Adapter ersetzt.

thomigueh, 2018-12-28:
Ich nutze das JavaScript aus den Beispielen um Elevation und Azimut der Sonne zu berechen. Diese Schreibe ich in ein Datenwort meines S7.Adapters.
Ab und an werden dann anscheint zufällige Zahlen vom s7.Adapter zurück gegeben, die so auch in der S7-300 stehen (Siehe Bild), manchmal wird auch einfach  eine Null geschrieben.

Ich habe inzwischen auch mit Node.Red in die S7 geschrieben dort sind mir diese Fehler nicht aufgefallen.
Ist das Problem bekannt und gibt es einen fix?

[image]

Apollon77, 2018-12-28:
Wenn du debug log aktivierst kommt da noch mehr Info irgendwie?

MarkF77, 2018-12-29:
Bei mir genau die gleichen Fehler, sowohl bei INT als auch REAL.
Teils 0, teils sehr große positive oder negative Zahlen.

Im debug log steht nichts.

thomigueh, 2019-01-07:
Moin,
also bei mir ergab das debug log jetzt nichts sinnvolles, ab und an kamen mal solche meldung
`
2019-01-06 16:28:35.524  - debug: s7.0 system.adapter.admin.0: logging false
--
2019-01-06 16:28:41.007  - debug: s7.0 system.adapter.admin.0: logging true
2019-01-06 16:29:15.530  - debug: s7.0 system.adapter.admin.0: logging false
2019-01-06 16:33:37.175  - debug: s7.0 system.adapter.admin.0: logging true
2019-01-06 16:41:24.550  - debug: s7.0 system.adapter.admin.0: logging false
`
die auf den S7 Adapter schließen lassen, das war es dann aber auch schon. Diese Meldungen decken sich Zeitlich aber auch nicht mit den auftretenden beschriebenen Fehler.

Apollon77, 2019-01-07:
Diese Log Zeilen kommen wenn Du im Admin auf Logs klickst :-)

fu-zhou, 2021-01-25:
Kaum 2 Jahre um, schon gibt es eine Idee:
@thomigueh @MarkF77 
Ich hatte das auch: Im Script (Blockly) muss der Real-Wert "aktualisiert" nicht "gesteuert" werden. Nachdem ich das umgestellt hatte, hat alles funktioniert - keine Ausreißer mehr, sieht dann so aus:
[image]
Binärwerte werden aber gesteuert, dass es funktioniert:
[image]

Ich hoffe, dass der Fall hier damit geschlossen werden kann?!

Apollon77, 2021-01-25:
mit "aktualisiert" schreibst Du nur werte in den State die NIE an das Gerät gesendet werden. Das wird nur bei "gesteuert" gemacht.

fu-zhou, 2021-01-25:
Bei der Kommunikation mit der S7 werden die konfigurierten DBs (Datenbaustein) zyklisch gelesen/ geschrieben (definiert im Adapter in "Poll-Delay").  Es wird also ein Real-Wert in einem DB aktualisiert (alle Datenpunkte in einem DB sind in iobroker als Objekt abgebildet, z.B. s7.0.DBs.DB22.PV_Netzleistung_IST) und der wird dann so im nächsten Zyklus geschrieben. Das o.a. Problem ist definitiv so behoben, wie oben von mir beschrieben. Ich hatte das auch schon mal im Forum gepostet, allerdings unter Blockly: https://forum.iobroker.net/topic/33900/blockly-sporadisch-werden-falsche-werte-geschrieben
Warum mit "steuern" sporadische Ausreißer geschrieben werden - keine Ahnung, aber die kommen auch in der S7 an und werden vom Script erzeugt, ich hatte da mal eine Fang-Schaltung in der S7 gebaut.
Versucht "steuern" evtl. das Schreiben zum Gerät zu triggern und dann kommt im Fall der S7 ab und zu Müll an, nämlich dann, wenn Polling und "steuern" genau zusammenfällt?

fu-zhou, 2021-03-08:
@Apollon77 Ich habe jetzt festgestellt, dass die Werte zwar wunderbar in den ioBroker-Objekten im S7-Adapter aktualisiert werden und es zu keinen Ausreißern mehr kommt, in der S7 aber immer "0" ankommt, du hast also (wie kann es anders sein ;-)) Recht mit "... NIE an das Gerät gesendet...". Jetzt habe ich die Blockly-Skripte wieder auf "steuere" umgerüstet und da kommen die Werte in der S7 an, aber sporadisch kommen eben auch die Riesen-Ausreißer - s.u.
[image]
Was kann ich tun, um dem Ganzen auf die Spur zu kommen?

Apollon77, 2021-03-08:
Ehrlich keine Ahnung, ich habe keineS7 und von daher ist das für mich unbekannt. Wo kommen die werte denn her? Logge doch im Blockyl mal welche werte du schreibst ... dann musst du doch wissen woher sie kommen ...

fu-zhou, 2021-03-08:
Wie kann ich im Blockly loggen? Ich logge die Werte in sql, daher sehe ich ja die Ausreißer. Wie oben schon geschrieben liegt es nach meinen bisherigen "Recherchen" nicht an der S7, sondern die Werte werden vom Skript an die S7 gegeben...

fu-zhou, 2021-03-08:
Meinst du über "verbose"?

Apollon77, 2021-03-08:
Es gibt einen Log block glaube ich

Zerotwo3185, 2021-04-27:
Guten Abend ich habe das gleiche Problem das gelegentlich falsche Werte in DBs geschrieben werden wollte mal fragen ob es mittlerweile eine Lösung für das Problem gibt?

fu-zhou, 2021-05-02:
Bei mir ist das sporadisch immer noch so. Ich gebe einen Grenzwert an die S7, der nicht ganz so kritisch ist, da bin ich jetzt auf den Zeit-Trigger "jede Minute" gegangen, seither gibt es keine Ausreisser mehr bei diesem Objekt - stabil seit 17.04.2021. Dann habe ich einen anderen Wert, der zu S7 muss, der wird durch die Wertänderung getriggert und da kommt es immer wieder zu den berühmten Ausreissern, so 2-4 Mal am Tag, ich finde aber kein Muster zur gezielten Reproduktion.

Zerotwo3185, 2021-05-04:
Ok vielen Dank für die Antwort. mir ist aufgefallen, wenn ich mir z.b die Uhrzeit aus dem Iobroker jede Sekunde übertragen lasse entstehen sehr viele Fehler. wenn ich jedoch nur alle 5 Sek. übertrage entstehen so gut wie keine Fehler. Werde es auch mal mit wertebereichen programmieren.

Wolfling78, 2022-10-22:
Dieser Topic wurde zwar geschlossen, aber das Problem besteht bei mir nach wie vor - gibt es hierzu einen Workaround?

fu-zhou, 2022-10-23:
Leider nein, ich würde durch Testen gerne nach wie vor meinen Beitrag zur Behebung des Verhaltens leisten, aber da müsste mir jemand auf die Sprünge helfen, weil ich so gar keine Ahnung habe vom Programmieren und dem dazugehörigen Debuggen... beschreiben und Beobachtungen reproduzierbar machen kriege ich aber gut hin, denke ich.

fu-zhou, 2022-10-23:
Ich habe Zugriff auf noch eine S7 CPU neben meinem Produktivsystem. Ich würde die mal mit ins System nehmen und mit Werten beschießen und dann mit der Poll-Zeit vom Adapter spielen etc. was mir halt so einfällt...

Zerotwo3185, 2022-10-23:
Guten Tag 
Ich bin dieses Problem umgangen indem ich jede Sekunde die Werte vergleiche und bei einem unterschied lasse ich die Werte erneut übertragen.
[image]
[image]

fu-zhou, 2022-10-23:
@Zerotwo3185 Guter Ansatz, um das Problem in den Griff zu bekommen - probiere ich mal. Interessanter (oder glücklicher) Weise passieren die Ausreißer nur bei Werten, die ich mit Blockly an die S7 übergebe, egal, ob zeit- oder wert-getriggert (sind nur ein paar Rechenwerte). Werte, die ich über vis in die S7 Schreibe (z.B. Sollwert Raumtemperatur) sind stabil und reißen nicht aus. Hast du das bei dir mal beobachtet?

Zerotwo3185, 2022-10-23:
Da ich Vis nicht nutze, sondern nur WinCC mit eine comfort Panel 1200. Und auf einer virtuellen Maschine ein PC System für ein 2tes Panel. kann ich dazu leider nichts sagen. alle Werte werden bei mir durch Blockly Scripte übertragen

fu-zhou, 2022-10-24:
@Zerotwo3185 Ich würde Version 2 mal probieren, ist für mich einfacher nachzuvollziehen. Hier hast du Eine eunktion "Anzeige" erstellt und, so verstehe ich dich, die triggerst du jede Sekunde dann, oder? So lange steht bei mir der viel zu große oder zu kleine Wert aber gar nicht im Objekt, das ist eigentlich im nächsten Poll-Zyklus schon wieder korrigiert.

Zerotwo3185, 2022-10-24:
@fu-zhou Ja ich triggere jede Sekunde. bei mir bleiben die werte solange im DB falsch bis sie neu übertragen werden.
zu version_1 Ich erzeuge 2 leere Listen und hole mir mit dem ID Selektor alle Werte von allen Homatic IP Wandthermostaten und addiere diese.
Dann mache ich dasselbe mit den werten die im DB der SPS stehen. Und wenn  die summen der beiden Listen nicht übereinstimmen übertrage ich die Werte erneut. 
beide Versionen werden jede sekunde getriggert.

fu-zhou, 2022-10-26:
basierend auf dem Input von Zerotwo3185 habe ich ein Workaround gebaut, der bei mit bisher funktioniert, ich beobachte das jetzt aber auch erst seit 1 1/2 Tagen. Das Ganze sieht so aus:
[image]
Zuerst schreibe ich den Wert von PV.Netzleistung in einen DB (PV_Netzbezug=pos_Netzeinspeisung=neg)
Nachdem Schreiben wird geschaut, ob die Werte übereinstimmen und wenn nicht wird nochmal nachgeschoben. Seither habe ich in den mit SQL aufgezeichneten Objekten keine Ausreißer mehr...

fu-zhou, 2022-11-22:
Jetzt nach knapp 4 Wochen in Betrieb kann ich bestätigen, dass ich mit dem o.a. Workaround keine Ausreißer mehr hatte und habe...

ea7777777, 2023-10-19:
Hallo, 

ich schlage mich auch schon ein 3/4 Jahr mit dem Problem rum. Selbes Verhalten wie bei den oberen Posts. In meinem Fall transferiere ich Werte von einer PV und einer separaten Batterie, welche über Modbus eingelesen werden über den S7-Adapter in die S7. 

Durch Fangschaltungen habe ich den Fehler eingegrenzt. Der Fehler passiert an der Stelle, wo die Werte in die S7 geschrieben werden. Also im S7-Adapter. 

Erst funktionierte der Workaround bei mir gar nicht. Im S7-Adapter hab ich für bestimmte Werte, die nur in die S7 geschrieben werden das Lesen aus der S7 deaktiviert (Haken bei Abfrage raus). Mit dem Haken bei Abfrage drin, werden die Werte von der S7 zum IO-Broker rückgelesen. 

Der Workaround funktioniert daher nur bedingt. Mit dem Workaround läuft das dann so, dass der falsch in der S7-Steuerung gelandete Wert zurück in den IO Broker gelesen wird. Über den Vergleich vom Original Wert und rückgelesenen Wert wird dann die Übertragung nochmal angestossen. 

Der Falsche Wert ist also immer noch eine gewisse Zeit in der S7, Immer noch besser als ohne Workaround. Da bleibt der falsche Wert nämlich so lange in der S7 bis sich der Original Wert ändert.

Wie bekommen wir den Sche** Fehler aus dem S7-Adapter?!?

fu-zhou, 2023-10-21:
Ich würde echt gerne dabei helfen, kann aber auch nur die S7-Seite testen, weil ich von der anderen Programmiererei keine Ahnung habe. Der Adapter basiert ja auf Snap 7 (https://snap7.sourceforge.net), aber ich kann nicht beurteilen, ob es an diesem Modul liegt oder an der Übergabe vom ioBroker. Wie wollen wir weitermachen? Ich kann mal schauen, ob ich jemanden in meinem Umfeld mit Javascript-Kenntnissen finde...

ea7777777, 2023-10-26:
Servus,

wäre toll, wenn der Bug endlich mal raus kommt. Es kann sein, dass er in Snap7 oder der Übergabe liegt, da hast du recht. Ich teste die nächsten Tage auch noch ein bisschen rum, um die Sache noch etwas einzugrenzen.

Damit ich das mal verstehe, wie kommst zu über Github zu dem Adapter, wenn du an dem eigentlich gar nicht´s programmierst. Was ist mit den ursprünglichen Programmierern... Apollo77, Bluefox...?

Gruß
Christian

fu-zhou, 2023-10-26:
Apollon77 und Bluefox sind gar nicht die ursprünglichen Programmierer, sondern (die) ioBroker-Entwickler. Sie haben den Adapter, ich denke mal basierend auf Sentry-Rückmeldungen, immer im ioBroker lauffähig gehalten, aber an der Schnittstelle zur S7 aus Mangel an Testmöglichkeiten nichts geändert bzw. die Issues hier aus Github behoben, wenn im Code die Problembeschreibung nachvollziehbar war = meine Version der Story ;-).

Das Ganze ist losgegangen mit smiling_jack, mit dem ich vor ca. 8 Jahren intensiven Kontakt hatte, weil ich da eben (schon) interessiert war, die S7-Seite mit dem Adapter zu testen und Ideen zur Funktionalität einbringen konnte. Die erste Anleitung zum Adapter kam von mir. Leider ist er dann irgendwann ausgestiegen und den hier beschriebenen Fehler habe ich damals noch nicht erkennen können, weil ich noch nicht im Produktivbetrieb war und die Ausreißer ja nur auffallen, wenn man die Werte aufzeichnet. Und, wie oben schon geschrieben, ist das Verhalten (bisher) nur erkennbar, wenn Werte über Skript geschrieben werden, über Vis (z.B. Sollwertvorgabe) kommen die Werte sauber an, da ist mir noch nie ein Ausreißer aufgefallen.

MarkF77, 2023-10-27:
Moin zusammen,
die Probleme treten auch bei Integer auf, nicht nur Real. Ich hatte als ursprünglich workaround versucht Real Zahlen (mit zwei Stellen nach dem Komma)  mit 100 zu multiplizieren, als Ganzzahl INT zu übertragen und in der S7 wieder durch 100 zu teilen. Auch bei INT dann teils extrem hohe Werte in der S7.
Es ist dabei auch egal ob es sich um einen Wert aus der Vis oder Javascript handelt. Über VIS fällt das natürlich weniger auf, da hier wesentlich seltener Werte gesendet werden 
Mit node-red treten diese Probleme nicht auf. Somit scheint es auch kein Problem von Snap7 zu sein.
Ich habe über logging schon alles mögliche versucht und nach meinen Erkenntnissen tritt der Fehler beliebig und nicht reproduzierbar auf.

Apollon77, 2023-10-27:
Hm ... also selbst theoritisch ... finde hier gerade kein Debug log das man ggf mal schauen könnte ob es was offensichtliches ist .. aber ja ... ich habe keine S7 ... also schwierig nachzustellen also geht es nur per lohgs

smolo-de, 2023-10-27:
Ich hatte 2020 schon diverse Konvertierungsprobleme im Adapter und zur S7-1500 festgestellt. Ich hab das zwar notiert bekomme das aber grad nicht mehr zusammen. Was ich damals festgestellt hatte war, das durch die Konvertierungen / Übertragung in String teilweise die Werte verfälscht werden. Das Zurückschreiben zum Adapter führt dann wieder zu kaputten Daten etc.. Das liegt zum Teil daran das die Typen nicht zusammenpassen bzw. der Inhalt String 254 zu 255 Zeichen und ASCII Zeichensätze. Leider habe ich keine Zeit mehr gefunden das auszuarbeiten :-(

ea7777777, 2023-10-27:
@Apollon77 Ich hätte hier schon ne S7 319 rum liegen, die ich dir zum Testen zuschicken könnte. Beim Testen würde ich dir auch gerne soweit helfen, bis wir den Fehler weg haben. Er nervt mich wie schon gesagt seit nem 3/4 Jahr.

Apollon77, 2023-10-27:
Danke Euch zweien ... Das Thema ist das ich seit fast einem Jahr an Matter für iobroker arbeite und gerade da sehr tief drin bin. Mit allem anderen nebenher (und hier und da Alexa Ablenkungen) schafft mein gehirn es gerade nicht noch ein Thema tief anzuschauen. von daher muss ich euch vertrösten aber es bleibt auf der Liste.

Wenn natürlich Infos wie von @smolo-de  zusammengefasst vllt das problem zeigen dann ist eine Code Anpassung (zusammen mit ggf Debug log gar nicht so schwer). Von daher wäre das das was aktuell mehr hilft als die Hardware fürchte ich :-(

Danke für Euer verständnis.

ea7777777, 2023-10-27:
Matter brauchen wir sicher alle früher oder später. Also sehr geil, dass du daran arbeitest! 

Wir können wirklich versuchen, den Fehler mit Debugs usw. zu dokumentieren, bis du Zeit findest. @smolo-de Vieleicht macht es sinn, uns da mal zusammen zu tun.... 

@Apollon77 Wenn du den Bug findest, schick ich dir jedenfalls ne Schnapsflasche deiner Wahl!

smolo-de, 2023-10-27:
Ja das wäre schon eine Sache aber ich brauche noch paar Monate, hänge grad mitten im Bauprojekt. Meine Anlage muss dann aber noch in den IOB integriert werden. Ich mache darüber auch mit Dali die ganze Lichtsteuerung daher ist das mit den richtigen Konvertierungen bei mir auch ein echtes Thema speziell Zeitwerte, Zahlwerte etc.. Es gibt auch geringefügige Unterschiede bei der 300er und 1500er die ich im Einsatz habe. Sobald es bei mir was neues gibt würde ich mich melden.

ea7777777, 2023-11-05:
Ich hab noch ein bischen rumprobiert. 

Ich schiebe die Daten mit JS von den Modbus Adaptern in den S7 Adapter. Im JS habe ich eine Überwachung, wenn die Tags unplausible Werte annehmen. Diese schlagen aber nie an. Die selben überwachungen habe ich auf S7 Seite. Da schlagen sie an. 

Weiter oben im Thread wurde angenommen, dass der Fehler nur mit JS oder Blockly passiert und nicht mit NodeRed. Deshalb habe ich mal die Werte mit NodeRed statt mit JS in den Adapter schieben lassen. Die Fehler treten immer noch auf. 

Ausserdem hab ich mir mal die Werte genauer angesehen. In meinem Fall Int und Dint. Real habe ich nicht dabei. Die Fehler passieren mir bei allen sich häufig änderen Werten bzw häufig übertragenen Werten. Egal welchen Sollwert sie haben (auch bei dauerhafter 0, wenn ich den Wert jede sec rüberschiebe). 

Ich habe mal die Protokollebene vom S7-Adapter auf "Silly" gestellt. Der Adapter protokolliert dann alle gesendeten Werte, welche in die S7 geschrieben werden. Wenn ein Ausreisser passiert, dann findet man diese aber nicht im Protokoll. Dort ist anscheinend noch der richtige Wert vorhanden. Fehler meldet der Adapter auch nicht. 

Wenn ich mir die Werte Bitweise anschaue, kann ich keinen Zusammenhang sehen. Es sind wirklich zufällige positive oder negative.

Hier mal ein paar Beispiele:

Int Werte:
Istwert                  Sollwert                 
32809                  ca. 100
53453                  ca. 100
12410                  ca. 100
9494                    ca. 100

DInt Werte:
Istwert                  Sollwert        
-1337028859      ca. 200  
-2143383291      ca. 100
1492702470        ca. 1200
12364                  ca. 100

MarkF77, 2023-11-05:
@ea7777777 
das mit NodeRed hast du falsch verstanden. Der Fehler tritt nicht auf, wenn du mit NodeRed direkt in die S7 schreibst - also den S7 Adapter umgehst. Hierzu musst du in NodeRed eine eigene S7 Verbindung herstellen.
LG

ea7777777, 2023-11-05:
Hallo Mark, ich wusste gar nicht, dass es eine S7-Node gibt. Hab gleich mal die relevanten Werte so umgesetzt. Wäre schön, wenn es da wirklich nicht auftritt.

Zerotwo3185, 2023-11-10:
Hallo könntet ihr mir vielleicht sagen wie ich die variablen in node red konfigurieren muss? ich bekomme es nicht hin mir einen INT oder REAL wert zu übertragen. String und bool funktioniert.

ea7777777, 2023-11-10:
Servus,

die Anleitung ist eigentlich recht gut:

https://flows.nodered.org/node/node-red-contrib-s7

Musst hald statt DB5,X0.1 für Int DB42,I3 oder für Real DB21,R7 Real adressieren. Dies klappte bei mir problemlos. Wenns nicht klappt mal mit einer Debug Node vor der S7 Node nachschauen, was drinnen steht. Oder im Info fester rechts schauen, ob er irgend einen Fehler hat.

Gruß
Christian

fu-zhou, 2023-11-10:
muss ich mir mal anschauen, gibt es die Möglichkeit, die Objekte anlegen zu lassen? Ich habe in der S7 Instanz nämlich 773 Objekte, dich ich als TSV einfach bei S7 reinkopiere...

ea7777777, 2023-12-09:
Servus zusammen,

inzwischen bin ich mit all meine Tags vom S7 Adapter nach NodeRed umgezogen. Die Werte werden ohne fantasie Zahlen übermittelt. Die Strings funktionieren nun problemlos bis zur Länge 254...  Was soll ich sagen... es funktioniert alles tadellos. Das schöne an NodeRed ist auch, dass man seine aktuellen Werte sieht und die Variablen gleich untereinander verknüpfen kann. Extra Java oder Blocky Scripte kann ich mir daher auch sparen.

Der S7 Adapter ist für mich daher komplett überflüssig geworden.

fu-zhou, 2023-12-10:
@ea7777777 interessant... Würdest du mal eine Anleitung posten, wie du automatisiert die Objekte angelegt hast? Davor graut mir nämlich, das für 773 Objekte händisch zu machen (s. mein Post vom 10.11.). Hast du auch Taster? Die sind im Adapter ja konfigurierbar. Oder würdest du ein Tasten über Node-Red realisieren (also Ausgang setzen und nach 200, 300 ms oder welche Zeit auch immer zurücksetzen = Ansteuerung Stromstoßrelais)?

ea7777777, 2023-12-10:
Nix automatisiert.... hab ca. 350 Variablen in Handarbeit rübergepackt. Bei mir ist die ganze Logik in der S7 Steuerung. Der IOBroker hat nur die Aufgabe, eine Schnittstelle zu Geräten zu schaffen, welche ich nicht direkt mit der S7-319 ansprechen kann. 

Die S7-Tags kann man in eine Excel Tabelle packen und dann importieren. Der Rest ist Handarbeit.

fu-zhou, 2023-12-10:
Ich brauche auch "nur" die Variablen meiner Kommunikations-DBs als Objekte für Node-Red, der Rest läuft in der S7. Im Adapter kann man das ja wunderbar im richtigen Format als TSV importieren und dann ist alles da als Objekte. Wie hast du die S7-DBs in Excel gepackt und wo/ wie importiert? Was ist dann genau die Handarbeit? Im Node-Red dann die importierten Variablen anbinden (das wäre logisch)?

Zerotwo3185, 2023-12-10:
Ich habe es aus dem s7.adapter in eine Excel Tabelle exportiert und dann mit Chat GPT alles in das Format gebracht das node Red benötigt .

fu-zhou, 2023-12-10:
Okay, die Langeweile über die anstehenden Feiertage wird sich somit in Grenzen halten... Wollen wir mal im ioBroker-Forum unter Node Red "Kommunikation mit S7" starten und da die Erkenntnisse sammeln. Beginnend bei "npm install node-red-contrib-s7"?

fu-zhou, 2023-12-10:
> Ich habe es aus dem s7.adapter in eine Excel Tabelle exportiert und dann mit Chat GPT alles in das Format gebracht das node Red benötigt .

Hat Chat GPT dann ein Import-File für den Flow rausgespuckt, so dass "s7 in" und "s7 out" nach Import schon verbunden ist?

Zerotwo3185, 2023-12-10:
Nein es hat mir nur das format der Excel tabellen umgewandlet so das ich dann in node red einfach die die tabellle wider einfügen konnte dann kann man ja die variablen im dropdown menü auswählen. Chat gpt hat aber ein paar anläufe gerbaucht bis es richtig funktioniert hat.

ea7777777, 2023-12-11:
@Zerotwo3185 
Würd mich auch interessieren wie du dies hinbekommen hast. Chat GPT hat doch keinen Dateiupload. Hast du den Excel File Quelltext direkt in die Texteingabe reingepackt?

Zerotwo3185, 2023-12-11:
Ja habe eine Excel Tabelle mit allen Datentypen als Beispiel erzeugt und anschließend meine Daten in eine Excel Tabelle eingefügt. Dann habe ich Chat gbt gesagt er soll meine Excel Tabelle nach dem oberen Vorgabe Konvertieren. Danach sollte Chat GPT daraus 2 Spaten machen so das ich die Spalte mit den DBs und den Kommentaren Einzelteil kopieren konnte. Wie gesagt hat nicht auf direkt geklappt bräuchte so ca. 6-7 anläufe.

fu-zhou, 2023-12-11:
Ich war jetzt mal so frei, unter Node Red im Forum ein Them aufzumachen:
https://forum.iobroker.net/topic/70938/s7-mit-node-red-node-red-contrib-s7
Sollen wir uns da weiter austauschen?

Ich habe mal eine Variable manuell angelegt und die Liste dann exportiert, da war das Format dann klar...

fu-zhou, 2024-01-08:
Ich glaube, dass ich einen Schritt weiter bin, s. https://forum.iobroker.net/post/1109064

Zusammenfassung: Ich habe unter Umgehung des S7 Adapters in der Javascript-Umgebung von ioBroker direkt über node Snap7 in die S7 geschrieben (vielen Dank an @Bettman66) und das gleiche Verhalten festgestellt. Als ich für jeden zu schreibenden Wert einen eigenen Buffer angelegt habe, war das Problem verschwunden!

Jetzt die Frage: Wer kennt sich mit Javscript und dem Buffer aus und kann den S7 Adapter entsprechend anpassen?

fu-zhou, 2024-01-10:
@bettman66 hat im Adapter eine klitze kleine Kleinigkeit verändert und seither (gestern Abend) läuft das Ding stabil ohne Ausreißer!!!
Es wäre schön, wenn Ihr den Adapter mal von Github (https://github.com/Bettman66/ioBroker.s7) installiert und das Verhalten über die nächsten Tage beobachtet und entweder hier oder im Forum (hoffentlich) bestätigt, dass die Ausreißer auch bei Euch nicht mehr vorkommen.

Vielen Dank vorab!

closed 2024-02-17 by Apollon77