ioBroker/ioBroker.s7 #17: Export des DB´s nicht möglich, Scrollbalken

HansPeterMeier, 2019-06-24:
Guten Tag,
ich habe ein Problem mit dem S7 Adapter. Wenn sehr viele Datenpunkte angegeben sind, ist es nicht möglich bis nach unten zu Scrollen um den Export durchzuführen.
Die Funktion "In Zwischenablage" scheint auch nicht zu funktionieren, wäre mir da aber noch egal, wenn ich bis zum letzten Datensatz scrollen könnte.
Dieses tritt beim Chromium, Iexplorer, Firefox, Safari auf.
Adapter ist 1.1.4.

Gruß und Danke

[image]

fu-zhou, 2020-04-26:
Muss ich leider auch feststellen: Leider funktioniert der CSV-Export der DBs im S7 Adapter nicht. Es kommt zwar die Meldung "Text wurde in die Zwischenablage kopiert", da ist dann aber nichts, was z.B. im Text-Editor eingefügt werden kann, um die Datenpunkte zu sichern. V1.1.7

ambo11, 2020-08-01:
gleiches Problem - der Text wird nicht in die Zwischenablage kopiert und man kann ihn leider auch nicht auswählen. - V1.1.8

closed 2021-05-19 by stale[bot]