ioBroker/ioBroker.s7 #22: S7 1200 connection problem

karesz25, 2020-02-03:
Hello.
I just find a mistake in the code (main.js)
        main.acp.rack  = parseInt(main.acp.rack,  10)  || 0;
        **main.acp.slot   = parseInt(main.acp.slot,  10)  || 2;**
In case of S7-1200 PLC you have to use rack and slot number 0 but it is not possible to set up in the current version, so I modified and it is working now.
        main.acp.rack  = parseInt(main.acp.rack,  10)  || 0;
        **main.acp.slot   = parseInt(main.acp.slot,  10)  || 0;**
BR.

Apollon77, 2020-02-03:
A line number would have also helped ;-) I fixed it differently ...

BKaresz, 2020-02-03:
> A line number would have also helped ;-) I fixed it differently ...

Hello.
Line 545.
How you fixed?
Bye.

Apollon77, 2020-02-03:
fixed by https://github.com/ioBroker/ioBroker.s7/commit/e1b5daceae98aa3ec0031540b9517c5ccf16c3f5

Please try 1.1.7 from GitHub and give Feedback

Apollon77, 2020-02-18:
Any success??

BKaresz, 2020-02-19:
Sorry I didn't try your solution yet, may be in the weekend. Thanks.

BKaresz, 2020-03-08:
Hi.
Sorry for late feedback, but I have no time..
I tried and it is working.
Thx.

closed 2020-03-08 by Apollon77