ioBroker/ioBroker.s7 #28: Nach Update auf JSC 3.0.20 Warnmeldungen im LOG

Steff42, 2020-04-29:
Hallo,
seit dem Update auf den JSC 3.0.20 bekomme ich folgende Warnmeldung(en) im LOG
[image]
Gruß Steffen

Steff42, 2020-04-29:
Habe mir mal das JSON eines der Objekte, das angewarnt wird angeschaut.
[image]
Das Problem kommt daher, das dort eine Zahl steht.
Gruß Steffen

Apollon77, 2020-05-01:
Jupp da empfehlt ein .toString()

Steff42, 2020-05-02:
Gerade gesucht, gefunden und getestet!
S7-Adapter gestoppt.
In der main.js jeweils in den Zeilen 771 (Inputs), 809 (Outputs) und 847 (Markers) ein .toString() angehängt.
S7-Adapter gestartet und die Warnungen sind weg.

Apollon77, 2020-05-02:
Kannst du einen PR machen dazu? Ansonsten reichen diese super Infos for mich die Tage auch. Danke

Steff42, 2020-05-02:
Einen PR habe ich noch nie gemacht.
Habe mich mit den vielen Funktionen im Github noch nicht wirklich auseinandergesetzt bzw. auch nicht verstanden wie das genau funktioniert und welche Voraussetzungen ich dazu benötige.
Sorry!

Steff42, 2020-05-03:
@Apollon77,
habe mich mal an einem PR versucht.
Hoffe das passt so!
Gruß Steffen

closed 2020-05-05 by Apollon77