ioBroker/ioBroker.s7 #37: V1.1.10: S7 Adapter bleibt "gelb" kommuniziert aber mit der S7

fu-zhou, 2021-01-25:
@smiling-Jack : schön, wieder was von dir zu hören! Ich habe heute gleich den Adapter aktualisiert, mit dem Ergebnis aus dem Titel. Was kann ich tun? (Unterstützung beim Debuggen etc.)
[image]

Kümmerst du dich auch um z.B. https://github.com/ioBroker/ioBroker.s7/issues/17 ?

Gruß!

ChrisM246, 2021-01-25:
Habe leider das gleiche Problem, die Verbindung scheint aber trotzdem zu funktionieren. Ein Einfacher Anzeigefehler vielleicht?

Bettman66, 2021-01-25:
Ich schließe mich an.

Apollon77, 2021-01-25:
Bitte GitHub version (bzw morgen latest 1.2.1) versuchen

fu-zhou, 2021-01-26:
1.2.2 funktioniert wieder!

closed 2021-01-26 by fu-zhou