ioBroker/ioBroker.s7 #38: Adapter meldet falschen Connection Status

MeinzIsAs, 2021-01-25:
|Systemdata| Bitte Ausfüllen|
|--|--|
|Hardwaresystem:| ACEPC T11|
|Arbeitsspeicher:| 4GB |
|Festplattenart:| SSD |
|Betriebssystem:| Ubuntu 18.04.2 LTS |
|Node-Version:| 12.19.0 |
|Nodejs-Version:| 12.19.0 |
|NPM-Version:| 6.14.8 |
|Installationsart:| Skript |
|Image genutzt:| Nein |

Seit dem Udpate auf 1.1.10 des Siemens S7 - Adapter wird der Connection Status nicht mehr richtig geschrieben. Der Status wird bei einem Adapterneustart auf "false" gesetzt, aber nicht mehr auf "true". Wenn ich den Status selbst auf "true" setze, bleibt er true. Die Werte werden normal aktualisiert und die Instanz ist "Grün".

Danke für Eure Hilfe.

Apollon77, 2021-01-25:
Bitte 1.2.1 versuchen

MeinzIsAs, 2021-01-26:
Gerade die V 1.2.2 installiert. Sieht gut aus mit dem Status. Wird wieder geschrieben. 

Besten Dank

closed 2021-01-26 by MeinzIsAs