ioBroker/ioBroker.s7 #49: Seit JSC Update Warnung "is missing the required property "common.type"

MeinzIsAs, 2021-04-30:
Seit dem Update auf den JSC 3.3.1 wird mein Logfile mit Warnungen gefüllt.

z.B.: 

(3601) Object of state "s7.0.DBs.DB125.Temperatur_Puffer_Oben_SPS" is missing the required property "common.type"

[image]

In den RAW-Eigenschaften der Variable steht das hier:
`{
  "from": "system.adapter.s7.0",
  "user": "system.user.admin",
  "ts": 1619768929158,
  "common": {
    "name": "",
    "role": "value.temperature",
    "unit": "",
    "read": true,
    "write": false
  },
  "native": {
    "cat": "db",
    "type": "REAL",
    "db": "DB125",
    "len": "",
    "dbId": 125,
    "address": 126,
    "offsetBit": 0,
    "rw": false,
    "wp": false
  },
  "acl": {
    "object": 1636,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator",
    "state": 1636
  },
  "_id": "s7.0.DBs.DB125.Temperatur_Puffer_Oben_Kessel",
  "type": "state"
}`

Könnte es sein, dass hier die Eigenschaft "type" fehlt?

**Versions:**  
 - Adapter version: 1.2.5
 - JS-Controller version: 3.3.1
 - Node version: 12.22.1
 - Operating system: Ubuntu 18.04.2 LTS

Vielen Dank

fu-zhou, 2021-05-02:
Hallo, ich habe das selbe Problem, habe die Log-Stufe vom S7-Adapter erst einmal auf "Error" gesetzt, weil pro Minute 1 MB ins Log-File geschrieben wurde (670 Datenpunkte). Die 670 Objekte habe ich über den DB-CSV Import angelegt, d.h. aus meiner Sicht müsste in der Adapter-Konfiguration die Spalte "common.type" ergänzt werden, so dass man sie je Datenpunkt/Objekt entsprechend definieren kann oder der Adapter müsste "common.type" basierend auf z.B. der Spalte "Typ" anlegen. Das händisch im Objekt-Dialog nachzubauen wäre irre...
@Apollon77 könntest du das übernehmen, "common.type" vom Adapter automatisch je Objekt erzeugen zu lassen? Entweder aus der Spalte "Typ" oder über eine neue Spalte "common.type". Ich weiß, du hast keine S7, aber du hast in letzter Zeit immer für die Kompatibilität des Adapters gesorgt - auf jeden Fall danke dafür.
[image]

GermanBluefox, 2021-06-23:
Fixed in 1.3.1

closed 2021-06-23 by Apollon77