ioBroker/ioBroker.s7 #58: Aufruf DBs-Ansicht leere Seite

tugsi, 2021-06-27:
Wenn ich unter Instanzen mir meine DB-Liste anschauen möchte, bleibt die Seite leer.
Ich habe 4 unterschiedliche DBs, wobei ich bei 2 DBs nur so rund jeweils 10 Datenpunkte habe, bei einem so rund 10o und mein größter hat so ca. 400 Datenpunkte.

Anscheinend kommt die neue GUI (1.3.2 habe ich) nicht klar mit der Datenmenge.
Ist auch egal ob ich die neue oder alte Adminoberfläche einschalte.

Apollon77, 2021-06-27:
Gibt es in der Browser-Konsole einen Fehler?

tugsi, 2021-06-27:
Jepp sorry, hier mal der Aufruf in Chrome:
`TypeError: Cannot read property 'toString' of null
    at TsvDialog.js:39
    at Array.map (<anonymous>)
    at TsvDialog.js:39
    at Array.forEach (<anonymous>)
    at TsvDialog.js:38
    at Nc (react-dom.production.min.js:262)
    at t.unstable_runWithPriority (scheduler.production.min.js:18)
    at Uo (react-dom.production.min.js:122)
    at Mc (react-dom.production.min.js:261)
    at gc (react-dom.production.min.js:243)
ss @ react-dom.production.min.js:216
n.callback @ react-dom.production.min.js:216
hi @ react-dom.production.min.js:131
ps @ react-dom.production.min.js:220
Tc @ react-dom.production.min.js:259
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Rc @ react-dom.production.min.js:252
gc @ react-dom.production.min.js:243
(anonymous) @ react-dom.production.min.js:123
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Ko @ react-dom.production.min.js:123
Go @ react-dom.production.min.js:122
Le @ react-dom.production.min.js:292
Jt @ react-dom.production.min.js:73
react-dom.production.min.js:123 Uncaught TypeError: Cannot read property 'toString' of null
    at TsvDialog.js:39
    at Array.map (<anonymous>)
    at TsvDialog.js:39
    at Array.forEach (<anonymous>)
    at TsvDialog.js:38
    at Nc (react-dom.production.min.js:262)
    at t.unstable_runWithPriority (scheduler.production.min.js:18)
    at Uo (react-dom.production.min.js:122)
    at Mc (react-dom.production.min.js:261)
    at gc (react-dom.production.min.js:243)
(anonymous) @ TsvDialog.js:39
(anonymous) @ TsvDialog.js:39
(anonymous) @ TsvDialog.js:38
Nc @ react-dom.production.min.js:262
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Mc @ react-dom.production.min.js:261
gc @ react-dom.production.min.js:243
(anonymous) @ react-dom.production.min.js:123
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Ko @ react-dom.production.min.js:123
Go @ react-dom.production.min.js:122
Le @ react-dom.production.min.js:292
Jt @ react-dom.production.min.js:73`

tugsi, 2021-06-28:
Version 1.3.3 installiert und klappt.
Danke!

closed 2021-06-28 by tugsi