ioBroker/ioBroker.s7 #62: 1.3.3: DB Konfiguration (zu) langsam

fu-zhou, 2021-07-04:
Hallo,

wenn man unter "DBs" einen einzelnen Datenpunkt über "+" hinzufügen möchte, wird das System wahnsinnig langsam, die einzelnen Felder zu befüllen wird zu einer echten Geduldsprobe. Bei der Eingabe der DB-Adresse passiert es hin und wieder sogar, dass das Dialogfenster vom Adapter einfach weiß wird und man einen Refresh machen muss. In der Version mit der alten GUI hatte ich die Probleme nicht.

Ich habe 671 Datenpunkte in 7 DBs angelegt, die Kommunikation funktioniert einwandfrei.

Im TSV Import-/Export-Fenster muss man jetzt die Adresse beginnend mit "DB..." eingeben, also z.B. DB80 8, bisher hat als Adresse "80 8" gelangt, da das Ganze ja im DB Dialog passiert ist. Kann man jetzt Eingänge, Ausgänge, Merker, DBs im selben TSV-Dialogfeld (gemischt) importieren? Dann macht es Sinn, die Adressen mit Kennung (DB, E, A, M) einzugeben. Wenn geplant ist, das so zu lassen, muss ich meine TSV-Quelle (Excel) entsprechend ändern, so dass das führende "DB" ergänzt wird.

Was kann ich tun, um an der Behebung der langsamen Eingabe mitzuwirken?

Apollon77, 2021-07-04:
Gibt es meldungen in der Browser Konsole?

fu-zhou, 2021-07-05:
Habe eben noch einmal probiert:

Zeile hinzufügen mit "+" geht relativ zügig (<5 Sek bis die Zeile sichtbar wird). Grundsätzlich sollte die Zeile ganz oben, also sofort sichtbar, auftauchen, das ist im Moment nur abhängig von der Sortierung der Fall. Da kann man schon mal verwirrt eine weitere Zeile anlegen, weil man denkt, es hat nicht geklappt und dann scrollt man runter und sieht beide neuen Einträge - das ist hier aber erst einmal sekundär.
[image]

Wenn ich dann ins Feld "Name" oder "Beschreibung" klicke, blinkt sofort der Curser darin. Wenn ich dann zu tippen anfange, passiert zunächst gar nichts, der Cursor hört auf zu blinken und nach 20-30 Sek kommt die Fehlermeldung in der Konsole:
`instrument.ts:129 [2021-07-05T19:15:27.337Z] ws connection error: CLOSED_NO_STATUS
(anonymous) @ instrument.ts:129
error @ socket.io.js:61
socket.onclose @ socket.io.js:128
instrument.ts:129 [2021-07-05T19:15:27.518Z] Start reconnect 0
socket.io.js:61 [2021-07-05T19:15:27.549Z] ws connection error: CLOSED_NO_STATUS
error @ socket.io.js:61
socket.onclose @ socket.io.js:128
socket.io.js:59 [2021-07-05T19:15:30.101Z] Start reconnect 0
instrument.ts:129 [2021-07-05T19:15:30.181Z] Try to connect
socket.io.js:59 [2021-07-05T19:15:30.250Z] Try to connect`
Aber sofort danach wird das Eingetippte im Feld "Name" oder "Beschreibung" sichtbar.
[image]

Wenn ich dann ins Feld "Adresse" klicke, blinkt der Cursor sofort, wenn ich dann mit "Backspace" die Adresse lösche, um die Neue einzugeben, wird das Adapterfenster nach ein paar mal Taste drücken weiß und in der Konsole kommt folgende Meldung:
`react-dom.production.min.js:216 TypeError: Cannot read property 'includes' of undefined
    at a.value (VM136 main.7eedbf69.chunk.js:1)
    at VM136 main.7eedbf69.chunk.js:1
    at Array.sort (<anonymous>)
    at Object.o.getSortedData (VM136 main.7eedbf69.chunk.js:1)
    at VM136 main.7eedbf69.chunk.js:1
    at oa (VM135 2.93ec12c4.chunk.js:2)
    at Wa (VM135 2.93ec12c4.chunk.js:2)
    at Vs (VM135 2.93ec12c4.chunk.js:2)
    at Ec (VM135 2.93ec12c4.chunk.js:2)
    at Oc (VM135 2.93ec12c4.chunk.js:2)
ss @ react-dom.production.min.js:216
n.callback @ react-dom.production.min.js:216
hi @ react-dom.production.min.js:131
ps @ react-dom.production.min.js:220
Tc @ react-dom.production.min.js:259
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Rc @ react-dom.production.min.js:252
gc @ react-dom.production.min.js:243
(anonymous) @ react-dom.production.min.js:123
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Ko @ react-dom.production.min.js:123
Go @ react-dom.production.min.js:122
(anonymous) @ react-dom.production.min.js:244
Ne @ react-dom.production.min.js:292
Fe @ react-dom.production.min.js:49
(anonymous) @ react-dom.production.min.js:50
Mr @ react-dom.production.min.js:105
Zt @ react-dom.production.min.js:75
Qt @ react-dom.production.min.js:74
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Le @ react-dom.production.min.js:292
Jt @ react-dom.production.min.js:73
react-dom.production.min.js:123 Uncaught TypeError: Cannot read property 'includes' of undefined
    at a.value (VM136 main.7eedbf69.chunk.js:1)
    at VM136 main.7eedbf69.chunk.js:1
    at Array.sort (<anonymous>)
    at Object.o.getSortedData (VM136 main.7eedbf69.chunk.js:1)
    at VM136 main.7eedbf69.chunk.js:1
    at oa (VM135 2.93ec12c4.chunk.js:2)
    at Wa (VM135 2.93ec12c4.chunk.js:2)
    at Vs (VM135 2.93ec12c4.chunk.js:2)
    at Ec (VM135 2.93ec12c4.chunk.js:2)
    at Oc (VM135 2.93ec12c4.chunk.js:2)
value @ BaseRegisters.js:62
(anonymous) @ BaseRegisters.js:218
o.getSortedData @ BaseRegisters.js:214
(anonymous) @ RegisterTable.js:176
oa @ react-dom.production.min.js:157
Wa @ react-dom.production.min.js:180
Vs @ react-dom.production.min.js:269
Ec @ react-dom.production.min.js:250
Oc @ react-dom.production.min.js:250
xc @ react-dom.production.min.js:250
gc @ react-dom.production.min.js:243
(anonymous) @ react-dom.production.min.js:123
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Ko @ react-dom.production.min.js:123
Go @ react-dom.production.min.js:122
(anonymous) @ react-dom.production.min.js:244
Ne @ react-dom.production.min.js:292
Fe @ react-dom.production.min.js:49
(anonymous) @ react-dom.production.min.js:50
Mr @ react-dom.production.min.js:105
Zt @ react-dom.production.min.js:75
Qt @ react-dom.production.min.js:74
t.unstable_runWithPriority @ scheduler.production.min.js:18
Uo @ react-dom.production.min.js:122
Le @ react-dom.production.min.js:292
Jt @ react-dom.production.min.js:73`
Danach ist Feierabend und ich muss einen Refresh machen.
[image]

Ich hoffe, das hilft jetzt mal, wenn weiterer Input nötig ist, bitte melden. Danke vorab!

GermanBluefox, 2021-07-06:
Ich habe Edit-Verhalten geändert (auch import). Kannst du neue Version ausprobieren?

fu-zhou, 2021-07-06:
habe 1.3.4 installiert:
Feld "Name" und "Beschreibung" bringt beim Eintragen jetzt keinen Fehler mehr und ist wesentliche schneller, aber die eingegebenen Zeichen erscheinen erst nach ca. 2 Sek im Feld - ist aber akzeptabel, weil die Masse wird ja normalerweise über TSV importiert.
Beim Ändern der Adresse ist das Verhalten aber gleich geblieben: weißes Adapterfenster und Fehlermeldung in der Konsole:
[image]

GermanBluefox, 2021-07-07:
Bitte 1.3.5 probieren

fu-zhou, 2021-07-07:
jetzt funktioniert's, einziger Schönheitsfehler ist die Eingabeverzögerung in allen Feldern von ca. 2 Sek - aber wie oben schon geschrieben: im Moment akzeptabel.

Die Sortierung wirkt sofort in Echtzeit, d.h. wenn man Name oder Adresse ändert, springt die neue Zeile sofort an die entsprechende Stelle in der Tabelle - je nach Sortierung. Ist etwas gewöhnungsbedürftig aber auch nachvollziehbar.

fu-zhou, 2021-07-07:
Muss ich beim Import noch etwas testen?

> Ich habe Edit-Verhalten geändert (auch import)

Was hast du beim Import geändert?

GermanBluefox, 2021-07-08:
> Muss ich beim Import noch etwas testen?
> 
> > Ich habe Edit-Verhalten geändert (auch import)
> 
> Was hast du beim Import geändert?

Ich habe gemacht, dass DB automatisch hinzugefügt wird.

fu-zhou, 2021-07-08:
Ich mache für TSV Import einen neuen Issue auf, okay? Es gibt ein paar Kleinigkeiten...

closed 2021-07-08 by fu-zhou