ioBroker/ioBroker.s7 #69: Nach Update JS Controller auf v3.3 wird für alle Objekte eine Warnung "is missing the required property "common.type"" angemeckert

TMHEINE, 2021-08-06:
Hallo,
hab jetze den JS controller von 3.2 auf 3.3 upgedatet
Jetzt bekomm ich auf alle objekte eine Warnung (Instanz läut aber).
z.B.
**(3608) Object of state "s7.0.DBs.DB1000.StromWLanRel_1" is missing the required property "common.type"**

dann hab ich alle Objekte rausgelöscht und den Adapter neu gestartet.
Dann wieder die Objekte (Merker und DB´s) importiert.

Das hat leider auch nicht geholfen.

Wie bekomm ich diese Flut an Warnungen weg?

Danke vorab für eure Hilfe

Apollon77, 2021-08-06:
Which s7 version you use? Please check Latest verison ... update to stable for that version comes tonight

TMHEINE, 2021-08-06:
Actually I use version 1.2.5

TMHEINE, 2021-08-06:
I try update to 1.3.6

TMHEINE, 2021-08-06:
looks OK now. Thank you!

closed 2021-08-06 by TMHEINE