ioBroker/ioBroker.s7 #76: S7 Adapter empfängt falsche Datentypen

Stefan592, 2021-09-10:
Es wird eine Warnmeldung ausgeben das die falschen Datentypen empfangen werden. Siehe Screenshots.

Parametrierung ist meiner Meinung nach korrekt.

PLC:
[image]

Objects:
[image]

Instance config:
[image]

Log:
[image]

Apollon77, 2021-09-10:
Bitte betroffene Objekte mal löschen und adapter restarten. Meldungen danach weg?

Stefan592, 2021-09-10:
> Bitte betroffene Objekte mal löschen und adapter restarten. Meldungen danach weg?

Nein. Die Datenpunkte werden wieder neu angelegt und die Meldung erscheint wieder.

Apollon77, 2021-09-10:
Dann itte Logging mal Debug stellen und schauen ob mehr Log messages kommen

Stefan592, 2021-09-10:
Leider kommt selbst auf Log Stufe "Debug" oder "Alles" nicht viel mehr raus:

<html>
<body>

s7.0 | 2021-09-10 09:32:26.592 | info | Successfully connected in S7 mode
-- | -- | -- | --
s7.0 | 2021-09-10 09:32:26.583 | info | Connect in S7 mode to 0 / 1
s7.0 | 2021-09-10 09:32:26.484 | warn | This object will not be created in future versions. Please report this to the developer.
s7.0 | 2021-09-10 09:32:26.483 | warn | Object DBs.DB1.gateGardenIsClosed is invalid: obj.common.write has an invalid type! Expected "boolean", received "string"
s7.0 | 2021-09-10 09:32:26.437 | warn | This object will not be created in future versions. Please report this to the developer.
s7.0 | 2021-09-10 09:32:26.436 | warn | Object DBs.DB1.gateKatjaIsClosed is invalid: obj.common.write has an invalid type! Expected "boolean", received "string"
s7.0 | 2021-09-10 09:32:26.163 | info | starting. Version 1.3.6 in /opt/iobroker/node_modules/iobroker.s7, node: v12.22.5, js-controller: 3.2.16
s7.0 | 2021-09-10 09:32:25.658 | debug | Plugin sentry Initialize Plugin (enabled=true)

</body>
</html>

Stephan84, 2021-10-20:
Ich habe die gleichen Meldungen:

s7.0 | 2021-10-20 23:04:05.898 | info | Successfully connected in LOGO! mode
s7.0 | 2021-10-20 23:04:05.887 | info | Connect in LOGO! mode to 512 / 256
s7.0 | 2021-10-20 23:04:05.806 | warn | This object will not be created in future versions. Please report this to the developer.
s7.0 | 2021-10-20 23:04:05.805 | warn | Object s7.0.DBs.DB1.Q4 is invalid: obj.common.write has an invalid type! Expected "boolean", received  "string"
s7.0 | 2021-10-20 23:04:05.712 | info | starting. Version 1.3.6 in /opt/iobroker/node_modules/iobroker.s7, node: v12.22.7, js-controller: 3.3.18

Bei mir werden die Werte in ioBroker aber passend angezeigt (wenn der Logo Ausgang 1 ist, zeigt ioBroker true an).

Apollon77, 2021-11-07:
Ok I assume the new UI is responsible.Can yu please show the native part of "system.adapter.s7.0" object? I need to see whats in the values. Ideallay have some with "RW" and some without please. Or Poll yes/no

@GermanBluefox it seems that the UI wrotes booleans as strings in config ... also "Length" is written as String which casued another issue ... I will have a commit for that soon

GermanBluefox, 2021-11-08:
Please test v1.3.7

Apollon77, 2021-11-08:
1.3.7. will not fix anything ... only the "write no boolean" issue ... I have some more fixes soon

Apollon77, 2021-11-09:
Bitte GitHub version versuchen, nach Installation Adapter restarten.

closed 2021-11-13 by Apollon77