ioBroker/ioBroker.s7 #87: String from IoB to S7 does not work

smolo-de, 2021-11-07:
Transfer of string from IoB to S7 does not work with following error:

s7.0 | 2021-11-07 12:18:08.504 | error | (19312) Exception-Code: ERR_INVALID_ARG_TYPE: The "size" argument must be of type number. Received type string ('202')
s7.0 | 2021-11-07 12:18:08.503 | error | (19312) TypeError [ERR_INVALID_ARG_TYPE]: The "size" argument must be of type number. Received type string ('202')     at Function.alloc (buffer.js:370:3)     at send (/opt/iobroker/node_modules/iobro
s7.0 | 2021-11-07 12:18:08.502 | error | (19312) uncaught exception: The "size" argument must be of type number. Received type string ('202')

S7 1500 with db element with STRING[20]

I think the error can be located at this line https://github.com/ioBroker/ioBroker.s7/blob/ce5e84ba5f18d6b83a4172f32b935ec34bcc0077/main.js#L319 because there could be an api change in nodejs. There was a known error in PLCHome NodeRed Adapter.

https://github.com/PLCHome/node-red-contrib-ads/issues/19#issuecomment-487273600

`old: var stringbuf = new Buffer(val.toString().slice(0,convert.length-1) + '\0', 'binary')
wrong: var stringbuf = Buffer.alloc(val.toString().slice(0,convert.length-1) + '\0', 'binary')
new: var stringbuf = Buffer.from(val.toString().slice(0,convert.length-1) + '\0', 'binary')`

Apollon77, 2021-11-07:
And how the config looks for this? We could make sure ".len" is a number

smolo-de, 2021-11-07:
> 
> 
> And how the config looks for this? We could make sure ".len" is a number

PLC:
[image]

S7 Adapter:
[image]

Object:
[image]

The bug with length of 2 is known.

Apollon77, 2021-11-07:
Ja das "len" hab ich beu mir denke schin gefickt. kann ich später pushen ... das mit "rw und poll"bezüg sich auf das andere issue ...

Apollon77, 2021-11-09:
Bitte GitHub version versuchen, nach Installation Adapter restarten.

GermanBluefox, 2021-11-09:
Please test v1.3.9

smolo-de, 2021-11-09:
Perfect no more problems 👍 

Tested with S7 1515-2pn

closed 2021-11-09 by smolo-de