ioBroker/ioBroker.s7 #89: Step 7 / TIA

DevKev21, 2021-11-19:
Hallo,
bezugnehmend auf das Thema Industrie 4.0 bin ich als SmartFactory Beauftragter meiner Firma auf der Suche nach einer Visualisierung, welche ich frei erstellen und mit unseren SPSen (300,1200,1500), sowie OPC-UA verbinden und steuern kann. Nachdem wir viel ausprobiert haben, bin ich auf den ioBroker gestoßen. Den Adapter hab ich bereits installiert und verbunden (ist auch grün), jedoch bekomme ich es nicht hin Ein- und Ausgänge, sowie Merker einzupflegen. Wie muss ich da vorgehen?
LG Kev

GermanBluefox, 2021-11-19:
Bitte mich unter info@iobroker.net anschreiben. 
Mit normalen SPS (kein LOGO) ging alles.

jogibear9988, 2022-02-06:
Wenn du mit einer 1500er kommunizieren willst, musst du auf der SPS die Put/Get Kommunikation aktivieren. Auch dürfen die Bausteine auf der SPS nur Absolut und nicht Symbolisch sein. Auf Vollsymbolische DBs kann man nur über ein neues Protokoll zugreifen, welches im Moment nur von AGLink unterstützt wird, welches nicht in diesem Adapter verwendet wird.

DevKev21, 2022-02-07:
Das hatte ich dann nach etwas Recherche im Internet auch herausgefunden. Da wir sowieso auf jede SPS im Netzwerk über OPC UA Tugriff haben, habe ich mal den OPC UA Adapter ausprobiert. Ich kann damit auch auf jede SPS zugreifen. Jedoch nicht auf unseren hauseigenen OPC Server (KEPServer Ex der Fa. inray). Vom handelsüblichen UA Expert Client kann ich problemlos standardmäßig drauf zugreifen, nur nicht über den OPC-UA Adapter. Wenn wir das Problem gelöst bekommen, kann ich den Adapter für alle SPSen in unserem Betrieb einsetzen, zwecks Maschinenvisualisierung etc. 
Vielen Dank!

jogibear9988, 2022-02-07:
Die 1500er liefern auch einen Opc Server direkt auf der Steuerung den man aktivieren kann.
Habt ihr den von Kepware auch einen UA Server? Wir hatten von denen mal einen DA Server im Einsatz.

DevKev21, 2022-02-07:
Auf unseren 1200er und 1500er läuft auch der eigene OPC-Server. Für unsere 300er CPUs nutzen wir den OPC-UA Server KEPServerEx in Kombination mit den IBH Link-Adaptern. Der funktioniert auch sonst super.

closed 2023-05-09 by stale[bot]