ioBroker/ioBroker.s7 #93: DBRead error[DB 1:0 - 7]: code: 0xa006e

Basti149, 2021-12-03:
**Describe the bug**  
A clear and concise description of what the bug is.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: <1.3.10 als auch mit 1.3.6>
 - JS-Controller version: <3.3.21> 
 - Node version: <16.3.1> 
 - NPM 8.1.2
 - Operating system: <Ubuntu Linux>

**Additional context** 
 
I have the following error message that came several times today. As soon as the connection was disconnected. All data points also stop and are no longer updated. I am happy to make myself available and test the versions.

`<html><body>

2021-12-03 20:49:10.340 | info | Successfully connected in LOGO! mode
-- | -- | --
s7.0 | 2021-12-03 20:49:08.882 | info | Connect in LOGO! mode to 512 / 512
s7.0 | 2021-12-03 20:48:18.882 | error | try reconnection: Disconnect successful: true
s7.0 | 2021-12-03 20:48:18.882 | warn | Poll error Last-Error Info: 655470:  ISO : An error occurred during recv TCP : Connection timed out
s7.0 | 2021-12-03 20:48:18.882 | warn | Poll error count: 1 code: 0xa006e
s7.0 | 2021-12-03 20:48:18.881 | warn | DBRead error[DB 1:0 - 7]: code: 0xa006e

</body>
</html>`

Apollon77, 2021-12-03:
As the error states the TCP connection times out , so please check your system

Basti149, 2021-12-03:
These are my settings for the adapter.

[image]

Basti149, 2021-12-05:
I continued researching today. Possibly this is due to Proxmox because something is being written to the log file there at the same time. Just what's the mistake.

`Dec 05 14:16:25 pve kernel: e1000e 0000:00:1f.6 eno1: NIC Link is Down
Dec 05 14:16:25 pve kernel: vmbr0: port 1(eno1) entered disabled state
Dec 05 14:16:28 pve kernel: e1000e 0000:00:1f.6 eno1: NIC Link is Up 100 Mbps Full Duplex, Flow Control: Rx/Tx
Dec 05 14:16:28 pve kernel: e1000e 0000:00:1f.6 eno1: 10/100 speed: disabling TSO
Dec 05 14:16:28 pve kernel: vmbr0: port 1(eno1) entered blocking state
Dec 05 14:16:28 pve kernel: vmbr0: port 1(eno1) entered forwarding state
`

Apollon77, 2021-12-05:
Yes this could be ... in this case the network connection dies and so the connection too but it seems that the adapter is not getting any info.

If it is an intel nuc maybe check out https://forum.proxmox.com/threads/e1000-driver-hang.58284/page-4#post-302924

closed 2022-04-27 by stale[bot]