ioBroker/ioBroker.s7 #113: DB integrate from S7-1200 TiaPortal

Saturdaynight80, 2022-03-30:
I need some support here.
I have installed the S7 adapter and it works very well.
Inputs and outputs as well as Marks can be exchanged wonderfully.
But unfortunately I do not understand how I can integrate the DB.

I bind the DB as described, DB13 0.0 or DB13 0
there is an error in the connection.
but in objects DB appears a folder with the datapoint.
[image]

if i write DB13.Wohnwand_1   
   or "ioBroker".Rezept[0]          (this is also correct in the Tia Portal)
The connection is ok
but in the folder DB is not a subdirectory with a datapoint

HansPeterMeier, 2022-04-27:
Du musst den optimierten Datenbaustein ausschalten, dann tauchen auch die direkten Adressen auf. Kämpfe gerade selber damit einen großen DB zu importieren, nur leider scheint das nicht so ganz zu funktionieren.

Saturdaynight80, 2022-06-03:
> Du musst den optimierten Datenbaustein ausschalten, dann tauchen auch die direkten Adressen auf. Kämpfe gerade selber damit einen großen DB zu importieren, nur leider scheint das nicht so ganz zu funktionieren.

Moin HansPeter,
bist du in diesem Thema weiter gekommen? 

ich habe übrigens mal versucht, deinen Tipp zu folgen und habe den Optimierten Bausteinzugriff deaktiviert. Aber sehe keinen Unterschied. Wo kannst du die absolute Adresse sehen?

MaikB85, 2022-06-03:
Rechtsklick auf den Datenbaustein und dann auf Eigenschaften. Unter Attribute findest du die Option "Optimierter Bausteinzugriff". Den Haken weg nehmen.
Nach erneutem Übersetzen werden dir im Datenbaustein die Adressen angezeigt.

GermanBluefox, 2022-06-20:
@GermanBluefox add to documentation

HansPeterMeier, 2022-06-21:
> Moin HansPeter, bist du in diesem Thema weiter gekommen?
>
> ich habe übrigens mal versucht, deinen Tipp zu folgen und habe den Optimierten Bausteinzugriff deaktiviert. Aber sehe keinen Unterschied. Wo [...]
[Tia zu iobroker.xlsx](https://github.com/ioBroker/ioBroker.s7/files/8946384/Tia.zu.iobroker.xlsx)

Ich habe mal eine Excel zusammengebaut auf die schnelle mit meinen Kenntnissen.
Man könnte jetzt eine  Beobachtungstabelle erstellen und die Tabelle kopieren und in die Excel einfügen, dann sollte rechts alles rauskommen, was man über den TSV Editor einfügen kann.
Es sind nicht alle Datentypen getestet, da fehlt mir leider die Zeit.
In der letzten Zeile sollte man den Type und Länge beachten, da ich die Länge z.B. beim DEZ+- nicht ermitteln kann ob INT oder DINT.
Vielleicht hilft es den ein oder anderen.
[Tia zu iobroker wandeln.xlsx](https://github.com/ioBroker/ioBroker.s7/files/8946447/Tia.zu.iobroker.wandeln.xlsx)

smilie108, 2023-12-01:
@GermanBluefox 
1. UINT und UDINT gibt es noch nicht oder ?
2. Hi gibt es eigentlich keinen einfacheren weg für den Import von dbs ? Weil von Tia raus geht ja noch in exel aber egal wie ich das exel dann als tabulator getrennt Speichere muss ich es immer händisch nochmal durchgehen da ich es sonst nicht in das tsv kopiren kann bzw es dann nicht geht.
3. Der import von dbs in welchem format muss die datei sein weil er bringt mir die meldung updated aber es ist dann nichts drinn ? 

Danke im voraus
Wenn du was zum testen brauchst ich kenne mich mit der S7 tia alle versionen bzw S7 bzw s5 sehr gut aus programiere große Anlagen. Hab zu hause selbst ne 1500er am laufen und ne 300er

mcm1957, 2025-08-25:
@Apollon77 
could be closed I think

closed 2025-09-13 by GermanBluefox