ioBroker/ioBroker.s7 #137: S7 Logo in der Instanzansicht wird nicht mehr angezeigt

Steff42, 2022-07-18:
Hallo,

in der Instanzansicht wird das S7 Logo nicht mehr angezeigt.

[image]

da müsste normalerweise diese Logo angezeigt werden.

[image]

in der Dev-Konsole wird der folgende Fehler angezeigt:

`
GET http://192.168.0.55:8081/adapter/s7/S7.png 404 (Not Found)

Image (async)

Zs @ react-dom.production.min.js:262

Gs @ react-dom.production.min.js:260

Gs @ react-dom.production.min.js:260

Xc @ react-dom.production.min.js:259

bc @ react-dom.production.min.js:283

$n @ react-dom.production.min.js:281

bs @ react-dom.production.min.js:270

Ln @ scheduler.production.min.js:13

Gn @ scheduler.production.min.js:14
`

Das Logo hat den Dateinamen s7.png, im GET Befehl wird der Dateiname mit S7.png angegeben. Das "S" ist groß geschrieben, vielleicht ist das der Fehler?

Gruß Steffen

Bettman66, 2022-07-23:
Das kannst du in der io-package.json ändern.
"icon": "S7.png", 
in 
"icon": "s7.png",

Apollon77, 2022-07-23:
ja werde ich fixen

Steff42, 2022-07-27:
Na Super,
jetzt habe ich hier einen commit erstellt, in dem die Änderung von @Bettman66 drin ist, aber ich kriege das mit dem PullRequest nicht gebacken.

Steff42, 2022-08-02:
@GermanBluefox ,

Thx for merging. Works now as expected.

closed 2022-08-02 by Steff42