ioBroker/ioBroker.s7 #153: New S7-1500 S7-1200 data types

HerrJensen81, 2022-10-14:
Hello,

thank you for that nice adapter.
I tried to add my DB's but i struggled because not all used generic data types are available.
It seems that only the old S7 Classic data types are available.

Is it possible to add the missing data types?
If I can help please let me know.

Greetings

Jens

Apollon77, 2022-10-15:
Is there any link to some specs for the datatypes?

HerrJensen81, 2022-10-18:
Hallo Ingo,

 

ich habe die Daten aus der TIA Hilfe ausgedruckt aber die PDF ist zu groß.

Ich versuche das mal in 2 mails zu schicken.

 

In der _Overview ist die Liste aller Datentypen.

Ich habe dann nur die Details der relevanten Datentypen angehangen.

 

 

Mit freundlichen Grüßen

Jens Siebel-Späth

 
 

Gesendet: Samstag, 15. Oktober 2022 um 15:36 Uhr

HerrJensen81, 2022-10-18:
Hallo Ingo,

vielleicht klappt das ja so:

[_Overview.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814390/_Overview.pdf)
[DATE.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814364/DATE.pdf)
[DATE_AND_LTIME.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814365/DATE_AND_LTIME.pdf)
[DATE_AND_TIME.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814366/DATE_AND_TIME.pdf)
[DINT.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814367/DINT.pdf)
[DTL.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814368/DTL.pdf)
[DWORD.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814369/DWORD.pdf)
[INT.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814370/INT.pdf)
[LINT.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814371/LINT.pdf)
[LREAL.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814372/LREAL.pdf)
[LTIME.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814373/LTIME.pdf)
[LTIME_OF_DAY.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814374/LTIME_OF_DAY.pdf)
[LWORD.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814375/LWORD.pdf)
[REAL.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814376/REAL.pdf)
[S5TIME.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814377/S5TIME.pdf)
[SINT.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814378/SINT.pdf)
[STRING.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814379/STRING.pdf)
[TIME.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814380/TIME.pdf)
[TIME_OF_DAY.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814381/TIME_OF_DAY.pdf)
[UDINT.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814382/UDINT.pdf)
[UINT.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814383/UINT.pdf)
[ULINT.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814384/ULINT.pdf)
[USINT.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814385/USINT.pdf)
[WCHAR.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814386/WCHAR.pdf)
[WORD.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814387/WORD.pdf)
[WSTRING.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814388/WSTRING.pdf)
[BOOL.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814391/BOOL.pdf)
[BYTE.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814392/BYTE.pdf)
[CHAR.pdf](https://github.com/ioBroker/ioBroker.s7/files/9814393/CHAR.pdf)

Gruß

Jens

HerrJensen81, 2022-10-25:
Helfen die Beschreibungen?

Apollon77, 2022-12-29:
Generell ja ... Dann mal zur konkreten Frage: Welche Datentypen fehlen Dir denn speziell? Weil ginge ja auch darum das das jemand dann testen kann - oder kannst Du alle 28 testen ;-)

HerrJensen81, 2022-12-30:
Hallo, hatte garnicht mehr mit einer Rückmeldung gerechnet. Ich bin erst wieder ab dem 4.1. 23 zuhause dann kann ich schauen welcher Datentyp es genau war. Es war ein datum/Zeit Typ. Ichkann aber auch alle testen. Kann ja einfach einen DB mit allen Typen anlegen und testen. Wie funktioniert der Excel Import? Gibt es eine Möglichkeit aus TIA direkt nach Excel zu gehen oder muss man das alles neu tippen? Gruß Jens Einen guten Rutsch

Apollon77, 2022-12-30:
Kein Stress, bei über 500 Adaptern und ein paar leuten die ca. 200 davon betreuen kann das schon mal passieren. Wenn es aber gelabelt ist sollte das nie vergessen werden. Deine Import Frage kann ich nicht beantworten weil ich nicht weiss was TIA ist (und auch keine S7 selbst habe). AM besten mach nen export aus dem adapter und vergleiche wie es aussieht

Patt2k, 2023-05-02:
Hey
Ich habe genau das gleiche Prob wollte ein LREAL übergeben und ging nicht.
Wenn du willst kann ich die alle 28 Testen da ich eine TIA 1500 CPU habe.

feuh, 2023-08-07:
Hallo,
ich würde auch einige der neuen Typen benötigen.
Stehe auch gerne zum testen zur Verfügung.

fu-zhou, 2024-01-13:
Der Adapter nutzt node-Snap7 über die API
https://github.com/mathiask88/node-snap7/blob/master/doc/client.md#data-functions
D.h. es können wohl nur die in node-Snap7 vorgesehenen Datentyp genutzt werden. Node-Snap7 baut wiederum auf Snap7 auf:  https://snap7.sourceforge.net