ioBroker/ioBroker.s7 #169: S7 Adapter funktioniert nach Upgrade auf Nodejs 16.18.1 nicht mehr!

AV196911, 2022-11-16:
Hallo Team S7 Adapter,
bei mir funktioniert der Adapter nach dem Upgrade auf Nodejs 16.18.1 nicht mehr.
Die gefundenen Einträge zu diesem Thema haben mir leider noch nicht geholfen.
Daher versuche ich es auf diesem Weg.

Hier ist der Auszug aus dem Protokoll:

host.ioBroker-Pi | 2022-11-16 21:54:07.951 | info | "system.adapter.s7.0" disabled
-- | -- | -- | --
host.ioBroker-Pi | 2022-11-16 21:54:01.585 | info | Rebuild for adapter system.adapter.s7.0 not successful in 3 tries. Adapter will not be restarted again. Please execute "npm install --production" in adapter directory manually.
host.ioBroker-Pi | 2022-11-16 21:54:01.585 | error | instance system.adapter.s7.0 terminated with code 1 (JS_CONTROLLER_STOPPED)
host.ioBroker-Pi | 2022-11-16 21:54:01.584 | error | Caught by controller[0]: }
host.ioBroker-Pi | 2022-11-16 21:54:01.584 | error | Caught by controller[0]: ]
host.ioBroker-Pi | 2022-11-16 21:54:01.584 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/lib/binding/node-v93-linux-arm/node_snap7.node'
host.ioBroker-Pi | 2022-11-16 21:54:01.583 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/addon-build/default/install-root/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.583 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/addon-build/debug/install-root/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.583 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/addon-build/release/install-root/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.583 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/compiled/16.18.1/linux/arm/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.582 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/build/default/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.582 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/Release/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.582 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/out/Release/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.581 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/Debug/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.581 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/out/Debug/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.581 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/build/Release/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.580 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/build/Debug/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.580 | error | Caught by controller[0]: '/opt/iobroker/node_modules/node-snap7/build/node_snap7.node',
host.ioBroker-Pi | 2022-11-16 21:54:01.580 | error | Caught by controller[0]: tries: [
host.ioBroker-Pi | 2022-11-16 21:54:01.580 | error | Caught by controller[0]: at Module._compile (node:internal/modules/cjs/loader:1155:14) {
host.ioBroker-Pi | 2022-11-16 21:54:01.579 | error | Caught by controller[0]: at Object.<anonymous> (/opt/iobroker/node_modules/iobroker.s7/main.js:9:21)
host.ioBroker-Pi | 2022-11-16 21:54:01.579 | error | Caught by controller[0]: at require (node:internal/modules/cjs/helpers:103:18)
host.ioBroker-Pi | 2022-11-16 21:54:01.579 | error | Caught by controller[0]: at Module.require (node:internal/modules/cjs/loader:1057:19)
host.ioBroker-Pi | 2022-11-16 21:54:01.578 | error | Caught by controller[0]: at Function.Module._load (node:internal/modules/cjs/loader:868:12)
host.ioBroker-Pi | 2022-11-16 21:54:01.578 | error | Caught by controller[0]: at Module.load (node:internal/modules/cjs/loader:1033:32)
host.ioBroker-Pi | 2022-11-16 21:54:01.578 | error | Caught by controller[0]: at Object.Module._extensions..js (node:internal/modules/cjs/loader:1209:10)
host.ioBroker-Pi | 2022-11-16 21:54:01.578 | error | Caught by controller[0]: at Module._compile (node:internal/modules/cjs/loader:1155:14)
host.ioBroker-Pi | 2022-11-16 21:54:01.577 | error | Caught by controller[0]: at Object.<anonymous> (/opt/iobroker/node_modules/node-snap7/lib/node-snap7.js:8:45)
host.ioBroker-Pi | 2022-11-16 21:54:01.577 | error | Caught by controller[0]: at bindings (/opt/iobroker/node_modules/bindings/bindings.js:126:9)
host.ioBroker-Pi | 2022-11-16 21:54:01.577 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/lib/binding/node-v93-linux-arm/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.576 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/addon-build/default/install-root/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.576 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/addon-build/debug/install-root/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.576 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/addon-build/release/install-root/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.575 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/compiled/16.18.1/linux/arm/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.575 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/build/default/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.575 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/Release/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.575 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/out/Release/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.574 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/Debug/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.574 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/out/Debug/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.574 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/build/Release/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.573 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/build/Debug/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.573 | error | Caught by controller[0]: → /opt/iobroker/node_modules/node-snap7/build/node_snap7.node
host.ioBroker-Pi | 2022-11-16 21:54:01.573 | error | Caught by controller[0]: Error: Could not locate the bindings file. Tried:
host.ioBroker-Pi | 2022-11-16 21:54:01.572 | error | Caught by controller[0]: ^
host.ioBroker-Pi | 2022-11-16 21:54:01.572 | error | Caught by controller[0]: throw err;
host.ioBroker-Pi | 2022-11-16 21:54:01.571 | error | Caught by controller[0]: /opt/iobroker/node_modules/bindings/bindings.js:135
host.ioBroker-Pi | 2022-11-16 21:53:59.128 | info | instance system.adapter.s7.0 started with pid 2250
host.ioBroker-Pi | 2022-11-16 21:53:58.460 | info | "system.adapter.s7.0" enabled

Solltet ihr andere Infos brauchen würde ich die natürlich nachreichen.

Apollon77, 2022-11-16:
Hast Du das getan was da ganz oben steht bei den Meldungen??

> Rebuild for adapter system.adapter.s7.0 not successful in 3 tries. Adapter will not be restarted again. Please execute "npm install --production" in adapter directory manually.

Apollon77, 2022-11-18:
Then you should update the packages on your system "apt update && apt dist-upgrade" ... Please open a forum thraed for support on these kind of things

closed 2022-11-18 by Apollon77