ioBroker/ioBroker.s7 #244: Device connection status in adapter objects

MartiMan79, 2023-08-07:
Hi,

I'm using a script using array enums to update values between S7 and hmip.
I would like to stop my scripts when the PLC is not connected but this is not available in the objects list of the adapter. 
Only alive and connected but not connected to device.

Or am I missing something?

Thanks!

closed 2023-08-08 by Apollon77

MartiMan79, 2023-08-08:
@Apollon77 No there is only the alive and connection state of the adapter, not the device.
If the device is offline, both are still true.

[image]

Apollon77, 2023-08-08:
Please exactly look the object patch I posted !! you look in system.adapter.s7.0 which is a different path!!