ioBroker/ioBroker.s7 #287: Adapter (v1.4.0) lässt sich nicht konfigurieren

mrjeschke, 2023-12-03:
Die Konfigurationsseite lässt sich nicht aufrufen. (Browserfehler)
Debian Bookworm
Nodejs 18.19.0
js-controller 5.0.17
admin 6.12.6

`Error: Minified React error #321; visit https://reactjs.org/docs/error-decoder.html?invariant=321 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at Z (react.production.min.js:18:327)
    at t.useContext (react.production.min.js:22:269)
    at makeStyles.js:189:40
    at withStyles.js:50:21
    at bc (react-dom.production.min.js:167:137)
    at ys (react-dom.production.min.js:193:57)
    at bi (react-dom.production.min.js:294:275)
    at xl (react-dom.production.min.js:280:389)
    at gl (react-dom.production.min.js:280:320)
    at fl (react-dom.production.min.js:280:180)
    at cl (react-dom.production.min.js:271:88)
    at rl (react-dom.production.min.js:268:429)
    at y (scheduler.production.min.js:13:203)
    at MessagePort.T (scheduler.production.min.js:14:128)
(anonym) @ console.ts:40
us @ react-dom.production.min.js:189
n.callback @ react-dom.production.min.js:189
Dr @ react-dom.production.min.js:144
Hi @ react-dom.production.min.js:262
xi @ react-dom.production.min.js:260
Mi @ react-dom.production.min.js:259
(anonym) @ react-dom.production.min.js:283
yl @ react-dom.production.min.js:281
rl @ react-dom.production.min.js:270
y @ scheduler.production.min.js:13
T @ scheduler.production.min.js:14
react.production.min.js:18 Uncaught Error: Minified React error #321; visit https://reactjs.org/docs/error-decoder.html?invariant=321 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at Z (react.production.min.js:18:327)
    at t.useContext (react.production.min.js:22:269)
    at makeStyles.js:189:40
    at withStyles.js:50:21
    at bc (react-dom.production.min.js:167:137)
    at ys (react-dom.production.min.js:193:57)
    at bi (react-dom.production.min.js:294:275)
    at xl (react-dom.production.min.js:280:389)
    at gl (react-dom.production.min.js:280:320)
    at fl (react-dom.production.min.js:280:180)
    at cl (react-dom.production.min.js:271:88)
    at rl (react-dom.production.min.js:268:429)
    at y (scheduler.production.min.js:13:203)
    at MessagePort.T (scheduler.production.min.js:14:128)
Z @ react.production.min.js:18
t.useContext @ react.production.min.js:22
(anonym) @ makeStyles.js:189
(anonym) @ withStyles.js:50
bc @ react-dom.production.min.js:167
ys @ react-dom.production.min.js:193
bi @ react-dom.production.min.js:294
xl @ react-dom.production.min.js:280
gl @ react-dom.production.min.js:280
fl @ react-dom.production.min.js:280
cl @ react-dom.production.min.js:271
rl @ react-dom.production.min.js:268
y @ scheduler.production.min.js:13
T @ scheduler.production.min.js:14
 `

Apollon77, 2023-12-03:
@GermanBluefox Please have a look ... I just did "minor" dep upgrades in the react area ...

Apollon77, 2024-02-17:
please try 1.4.3

closed 2024-02-17 by Apollon77