ioBroker/ioBroker.s7 #364: Update Node.js

Alexhouse0815, 2024-07-22:
Leider funktioniert nach dem Update von Node.js der Adapter nicht mehr.

`2024-07-21 13:22:49.800	debug	Redis Objects: Use Redis connection: 127.0.0.1:9001
 
s7.0
2024-07-21 13:22:49.853	debug	Objects client ready ... initialize now
 
s7.0
2024-07-21 13:22:49.859	debug	Objects create System PubSub Client
 
s7.0
2024-07-21 13:22:49.861	debug	Objects create User PubSub Client
 
s7.0
2024-07-21 13:22:49.927	debug	Objects client initialize lua scripts
 
s7.0
2024-07-21 13:22:49.943	debug	Objects connected to redis: 127.0.0.1:9001
 
s7.0
2024-07-21 13:22:49.993	debug	Redis States: Use Redis connection: 127.0.0.1:9000
 
s7.0
2024-07-21 13:22:50.016	debug	States create System PubSub Client
 
s7.0
2024-07-21 13:22:50.018	debug	States create User PubSub Client
 
s7.0
2024-07-21 13:22:50.094	debug	States connected to redis: 127.0.0.1:9000
 
s7.0
2024-07-21 13:22:50.168	debug	Plugin sentry Initialize Plugin (enabled=true)
 
`

Apollon77, 2024-07-23:
Can you pleas eprovide more infos ... nodejs update from which to which version? More logs pelase ... these logs show nothing

Alexhouse0815, 2024-07-23:
Update nodejs 18.20.2 or 3 --> Node.js: v20.15.1
NPM: 10.7.0

`<html>
<body>

s7.0 | 2024-07-23 10:02:58.282 | debug | Redis Objects: Use Redis connection: 127.0.0.1:9001
-- | -- | -- | --
s7.0 | 2024-07-23 10:02:58.336 | debug | Objects client ready ... initialize now
s7.0 | 2024-07-23 10:02:58.339 | debug | Objects create System PubSub Client
s7.0 | 2024-07-23 10:02:58.341 | debug | Objects create User PubSub Client
s7.0 | 2024-07-23 10:02:58.399 | debug | Objects client initialize lua scripts
s7.0 | 2024-07-23 10:02:58.409 | debug | Objects connected to redis: 127.0.0.1:9001
s7.0 | 2024-07-23 10:02:58.452 | debug | Redis States: Use Redis connection: 127.0.0.1:9000
s7.0 | 2024-07-23 10:02:58.473 | debug | States create System PubSub Client
s7.0 | 2024-07-23 10:02:58.474 | debug | States create User PubSub Client
s7.0 | 2024-07-23 10:02:58.554 | debug | States connected to redis: 127.0.0.1:9000
s7.0 | 2024-07-23 10:02:58.615 | debug | Plugin sentry Initialize Plugin (enabled=true)
s7.0 | 2024-07-23 10:02:59.265 | info | starting. Version 1.4.3 (non-npm: ioBroker/ioBroker.s7#42e71b0364b3a7c75eba8f4da883d240b073d874) in /opt/iobroker/node_modules/iobroker.s7, node: v20.15.1, js-controller: 6.0.6
s7.0 | 2024-07-23 10:02:59.432 | info | Connect in LOGO! mode to 256 / 512
s7.0 | 2024-07-23 10:02:59.443 | error | Connection failed. Code #111

</body>
</html>`

Alexhouse0815, 2024-07-23:
what kind of logs exactly?

Apollon77, 2024-07-24:
Hm ... Error 111 ... sag mir jetzt spontan nichts, Frag vllt auch mal im Forum. Mal gerät (s7) neu gestartet?

Alexhouse0815, 2024-07-24:
Div. Sachen schon probiert. Im Forum wurde ich hierher verwiesen. :-)

Alexhouse0815, 2024-07-24:
Habe jetzt ioBroker neu aufgesetzt. Läuft wieder, warum auch immer.