ioBroker/ioBroker.s7 #381: snap7 was compiled against a different Node.js version using

Dettenrocks, 2024-09-12:
**Describe the bug**  
Error: The module '/opt/iobroker/node_modules/node-snap7/build/Release/node_snap7.node'

was compiled against a different Node.js version using

NODE_MODULE_VERSION 108. This version of Node.js requires

NODE_MODULE_VERSION 115. Please try re-compiling or re-installing

the module (for instance, using `npm rebuild` or `npm install`).

    at Module._extensions..node (node:internal/modules/cjs/loader:1586:18)

    at Module.load (node:internal/modules/cjs/loader:1288:32)

    at Module._load (node:internal/modules/cjs/loader:1104:12)

    at Module.require (node:internal/modules/cjs/loader:1311:19)

    at require (node:internal/modules/helpers:179:18)

    at bindings (/opt/iobroker/node_modules/bindings/bindings.js:112:48)

    at Object.<anonymous> (/opt/iobroker/node_modules/node-snap7/lib/node-snap7.js:8:45)

    at Module._compile (node:internal/modules/cjs/loader:1469:14)

    at Module._extensions..js (node:internal/modules/cjs/loader:1548:10)

    at Module.load (node:internal/modules/cjs/loader:1288:32) {

  code: 'ERR_DLOPEN_FAILED'

}

Node.js v20.17.0

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: v1.4.3
 - JS-Controller version: v6.0.11
 - Node version: Node.js v20.17.0
 - Operating system: iobroker.docker v10

**Additional context**  
Add any other context about the problem here.

Apollon77, 2024-09-13:
ideally the iobroker should fix this automatically ...  if not, follow the infos in error message ;-)

cd /opt/iobroker/node_modules/node-snap7
npm rebuild

undeat, 2025-05-14:
Hello,

i allready had posted that issue in an opther topic i can not find any more. 

npm rebuild and npm install had helped last time like it was told in the log text but every time i restart iob i have to rebuild it again. Thats very annoying. I run it on Windows and sometimes my server restarts for updates and so i allways get that problem. When i used an older Node.js version i had no problems.

Apollon77, 2025-05-14:
That seems to be some weird NPM effect. No clue why. 
YOu could try to backup config and states of the adapter, then fully delete the adapter via Admin (not only the instance!). Then reinstall

closed 2025-08-16 by Apollon77