ioBroker/ioBroker.s7 #407: Reading Outputs from extension module

mscifm, 2025-01-11:
**Describe the bug**  
I try to READ Outputs form Q2.x or Q3.y but I receive no value, reading Q0.n work well
**To Reproduce**  
Steps to reproduce the behavior:
configure OUTPUTs greater than Q1.n

**Expected behavior**  
Read values
**Screenshots & Logfiles**  

**Versions:**  
 - Adapter version: 1.4.3
 - JS-Controller version: 7.0.6
 - Node version: 18.19.0
 - Operating system: debian 12
**Additional context**  
Add any other context about the problem here.

Apollon77, 2025-01-13:
Pleaae provide a debug log

mscifm, 2025-01-16:
In the WebUI log system I found the following entries
```
<html>
<body>
<!--StartFragment-->
s7.2 | 2025-01-16 19:01:43.116 | info | Successfully connected in S7 mode
-- | -- | -- | --
s7.2 | 2025-01-16 19:01:43.096 | info | Connect in S7 mode to 0 / 1
s7.2 | 2025-01-16 19:01:42.833 | info | starting. Version 1.4.3 in /opt/iobroker/node_modules/iobroker.s7, node: v18.19.0, js-controller: 7.0.6
s7.2 | 2025-01-16 19:01:42.581 | debug | Plugin sentry Do not initialize Plugin (enabled=false)
s7.2 | 2025-01-16 19:01:42.538 | debug | States connected to redis: 127.0.0.1:9000
s7.2 | 2025-01-16 19:01:42.475 | debug | States create User PubSub Client
s7.2 | 2025-01-16 19:01:42.474 | debug | States create System PubSub Client
s7.2 | 2025-01-16 19:01:42.463 | debug | Redis States: Use Redis connection: 127.0.0.1:9000
s7.2 | 2025-01-16 19:01:42.437 | debug | Objects connected to redis: 127.0.0.1:9001
s7.2 | 2025-01-16 19:01:42.427 | debug | Objects client initialize lua scripts
s7.2 | 2025-01-16 19:01:42.397 | debug | Objects create User PubSub Client
s7.2 | 2025-01-16 19:01:42.396 | debug | Objects create System PubSub Client
s7.2 | 2025-01-16 19:01:42.393 | debug | Objects client ready ... initialize now
s7.2 | 2025-01-16 19:01:42.358 | debug | Redis Objects: Use Redis connection: 127.0.0.1:9001
host.IoTHome | 2025-01-16 19:01:39.857 | info | instance system.adapter.s7.2 in version "1.4.3" started with pid 14615

<!--EndFragment-->
</body>
</html>
```

nothing more is shown

ifmmsc, 2025-02-02:
Someone with the same issue here?