ioBroker/ioBroker.s7 #427: Not Working NPM 22.x

undeat, 2025-03-02:
I use this adapter and now i cannot use iob with npm 22.x. I still use 20.x but i am running into problems with Homebridge adapter. Can you make the adapter compatible, please? 

Thanks a lot for that great adapter!

undeat, 2025-03-03:
> I assume you mean Node.js 22.x?
>
> Also, what does "cannot use" mean? Any logs? Any infos?
>
>
>
> I assume you mean Node.js 22.x?
>
> Also, what does "cannot use" mean? Any logs? Any infos?

Hallo,

Du sprichst meine ich auch Deutsch oder?

Danke für Rückmeldung. Ich meine natürlich node.js sry. Ich könnte heute Abend mal Logs aufnehmen. Es war eine ganze Reihe an Übersetzungsfehlern und Exceptions, die kommen sobald ich den Adapter starte und node.js 22 verwendet wird. Sollte sich denke ich schnell nachstellen lassen.

undeat, 2025-03-04:
Today i have tried to install node.js 22 instead of 23. Then i found out that the S7 Adapter is working @ node.js 23

now i made the update at my iobroker instance i use in production. There it tried node.js 22 and i get following error again

<html>
<body>

host.Server(SmartHome) | 2025-03-04 10:42:37.550 | info | Rebuild for adapter system.adapter.s7.0 not successful in 3 tries. Adapter will not be restarted again. Please execute "npm install --production" in adapter directory manually.
-- | -- | -- | --
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | instance system.adapter.s7.0 terminated with code 1 (JS_CONTROLLER_STOPPED)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[1]: Node.js v22.14.0
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[1]:
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: }
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: code: 'ERR_DLOPEN_FAILED'
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at Module._compile (node:internal/modules/cjs/loader:1554:14) {
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at Object.<anonymous> (D:\SmartHome\node_modules\node-snap7\lib\node-snap7.js:8:45)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at bindings (D:\SmartHome\node_modules\bindings\bindings.js:112:48)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at require (node:internal/modules/helpers:136:16)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at Module.require (node:internal/modules/cjs/loader:1311:12)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at wrapModuleLoad (node:internal/modules/cjs/loader:220:24)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at TracingChannel.traceSync (node:diagnostics_channel:322:14)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at Function._load (node:internal/modules/cjs/loader:1108:12)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at Module.load (node:internal/modules/cjs/loader:1289:32)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: at Object..node (node:internal/modules/cjs/loader:1732:18)
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: the module (for instance, using `npm rebuild` or `npm install`).
host.Server(SmartHome) | 2025-03-04 10:42:37.549 | error | Caught by controller[0]: NODE_MODULE_VERSION 127. Please try re-compiling or re-installing
host.Server(SmartHome) | 2025-03-04 10:42:37.548 | error | Caught by controller[0]: NODE_MODULE_VERSION 115. This version of Node.js requires
host.Server(SmartHome) | 2025-03-04 10:42:37.548 | error | Caught by controller[0]: was compiled against a different Node.js version using
host.Server(SmartHome) | 2025-03-04 10:42:37.548 | error | Caught by controller[0]: Error: The module '\\?\D:\SmartHome\node_modules\node-snap7\build\Release\node_snap7.node'
host.Server(SmartHome) | 2025-03-04 10:42:37.548 | error | Caught by controller[0]:
host.Server(SmartHome) | 2025-03-04 10:42:37.548 | error | Caught by controller[0]: ^
host.Server(SmartHome) | 2025-03-04 10:42:37.548 | error | Caught by controller[0]: throw e;
host.Server(SmartHome) | 2025-03-04 10:42:37.547 | error | Caught by controller[0]: D:\SmartHome\node_modules\bindings\bindings.js:121

</body>
</html>

undeat, 2025-03-04:
I am Running iob on Windows 11. Manuel rebuild with "npm install --production" did solve the problem

Apollon77, 2025-03-04:
Yes ... then solved, right? Log was telling exactly to do that

closed 2025-03-04 by Apollon77